admonstrator/glinet-tailscale-updater #1: Tailscale security vulnerability

Aubermean, 2024-05-14:
With the latest version of Tailscale, you might need to add `--stateful-filtering=false`. See the release notes for v1.66.0

https://github.com/tailscale/tailscale/releases

Admonstrator, 2024-05-14:
This must be done by GL.iNet, the update script does not modify the startup scripts.

Aubermean, 2024-05-14:
GL.iNet does not (and likely will not) provide stable Tailscale support, it'll probably be a century before they update Tailscale, so I thought it would be smarter to report it here. It eluded me for ages today trying to figure out why suddenly everything was broken - hopefully might save someone some time. Thanks for the updater!

Admonstrator, 2024-05-14:
I'll talk to them to adjust the tailscale integration script.

Aubermean, 2024-05-14:
It's only relevant to the latest version, that flag isn't available in older versions. Tailscale did some server side mitigation too but it's unclear if that will have any effect on older versions. For now, they are just suggesting an upgrade. The risk in particular is to Linux and site to site networking.

Admonstrator, 2024-05-14:
Guess it's just Tunnelvision mitigation - which is a risk indeed; but not as bad as people think. 

I will think about a way to add the parameter to the script somehow.

Aubermean, 2024-05-14:
The parameter itself is basically disabling their clientside mitigation. Anyway whether or not it is ever actively exploited, the greater issue/frustration/risk is 24 hours of head scratching why suddenly your whole network isn't working properly :)

So I think even just a warning/explainer so people stumble via Google a solution sooner rather than later, is all the help anyone will need!

Aubermean, 2024-05-16:
For things to work as they used to, site-to-site stuff, in combination with an exit node (they support Mullvad natively) or acting as a exit node... it needs to be set. Without it things won't work. A warning or pointer to this fact, for 1.66 and up, is enough. Failing that hopefully this issue indexed on google will provide a solution. :)

Admonstrator, 2024-05-17:
I would assume you need to modify `/etc/init.d/tailscale` and add `procd_append_param command --stateful-filtering=false` within the `start_service()` function. But this needs to be tested.

Admonstrator, 2024-05-19:
A fix was integrated to modify the `gl_tailscale` script - so active exit nodes should work again.

Admonstrator, 2024-05-19:
I got a confirmation that it works now.

closed 2024-05-19 by Admonstrator