admonstrator/glinet-tailscale-updater #2: Feature Request: Provide pre-built and pre-compressed binaries

lwbt, 2024-06-08:
Hello, I see you have moved the script to its own repository and updated a few lines (like xz-utils is just xz now) after I suggested compressing with UPX.

I went a bit further and created a GitHub Actions workflow to compile a [Tailscale combined](https://tailscale.com/kb/1207/small-tailscale) binary from source and compress it with UPX.

https://github.com/lwbt/ts_build_test/blob/main/.github/workflows/build.yaml

The workflow runs every hour, which might be a bit much and it completes in less than 5 minutes (3?). Occasionally it throws an error which the owner of the account receives via mail. If I recall it's not more than once per week on average. If you reduce the frequency you will also reduce the likeliness of these errors to occur (they seem to be random connectivity issues and I had no time yet to make the workflow more robust).

Would you be willing to add this workflow file to your repository and point your updater script in the same repository to download the created releases?

I have also created a workflow for AdGuard Home https://github.com/lwbt/agh_build_test/blob/main/.github/workflows/build.yaml

A GitHub account on the free tier is just fine to run these Actions. I currently have 0 pipelines minutes spent this month, while running both every hour (they just run for 10 - 15s if there is nothing to do).

Admonstrator, 2024-06-08:
Thanks, I will add this feature in the next version!

Admonstrator, 2024-06-09:
Feature was added in 51e76795c13a3d46e6bb38c8e1c1e6318f2468d2

Admonstrator, 2024-06-20:
A bug was fixed where the official repository published a new release but forgot to move the _latest_ tag.

closed 2024-06-20 by Admonstrator