admonstrator/glinet-tailscale-updater #4: Last update.sh version broken instalation

ncosta24, 2024-06-14:
Hello,

First, thank very much for your work.

I run your script last version and don't realize than now you install tiny version.

The scrip run well but now tailscale is not working and when trying to execute tailscale start it gives a error "illegal instruction"

Admonstrator, 2024-06-14:
Could you please share some information about the device you run this script on?

ncosta24, 2024-06-14:
[image]

ncosta24, 2024-06-14:
https://www.gl-inet.com/products/gl-xe300/

Admonstrator, 2024-06-14:
Could you please run `uname -a` on this device?

ncosta24, 2024-06-14:
[image]

ncosta24, 2024-06-14:
> Could you please run `uname -a` on this device?
[image]

ncosta24, 2024-06-14:
with --no-tiny
[image]

Admonstrator, 2024-06-14:
Please try to run the script like `./update-tailscale.sh --no-tiny` to download the official tailscale version and see if that fix it.

Admonstrator, 2024-06-14:
Please run
```
wget -qO /tmp/tailscale.tar.gz https://pkgs.tailscale.com/stable/tailscale_1.68.1_mips.tgz
./update-tailscale.sh --no-download --no-tiny
```

ncosta24, 2024-06-14:
> Please run
> 
> ```
> wget -qO /tmp/tailscale.tar.gz https://pkgs.tailscale.com/stable/tailscale_1.68.1_mips.tgz
> ./update-tailscale.sh --no-download --no-tiny
> ```

[image]

ncosta24, 2024-06-14:
[image]

ncosta24, 2024-06-14:
Download the file from a private location. Then:
[image]

Admonstrator, 2024-06-14:
Thanks! Seems that the tailscale download server does not allow wget to download. I remember about this issue. 

Will try to fix it on sunday.

Admonstrator, 2024-06-19:
The issue is fixed in 343a8d9b953915e84f38e7f45b6f5360a1861ed0

ncosta24, 2024-06-19:
[image]

ncosta24, 2024-06-19:
[image]

Admonstrator, 2024-06-19:
Can you please test the download by running
```sh
wget -U "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/126.0.0.0 Safari/537.36" -qO /tmp/tailscale.tar.gz https://pkgs.tailscale.com/stable/tailscale_1.68.1_mips.tgz
```

ncosta24, 2024-06-19:
[image]

Admonstrator, 2024-06-19:
Are you using VPN or something?

ncosta24, 2024-06-19:
4g connection on the router

Admonstrator, 2024-06-19:
The problem is that I can't figure out why dl.tailscale.com is blocking you. I already implemented a fix for the old wget version so a newer user agent will be sent ... but since you are still blocked it's an issue I can't fix on my side :(

ncosta24, 2024-06-19:
:(

Admonstrator, 2024-06-19:
If you use a PC, can you download the file `https://pkgs.tailscale.com/stable/tailscale_1.68.1_mips.tgz`?

ncosta24, 2024-06-19:
yes
And if i share it on my synology drive i can download it on the gl. inet router

Admonstrator, 2024-06-19:
Can you try
```
curl https://pkgs.tailscale.com/stable/tailscale_1.68.1_mips.tgz --output /tmp/tailscale.tar.gz -v 
```
please?

ncosta24, 2024-06-19:
[image]

ncosta24, 2024-06-19:
With curl, works

Admonstrator, 2024-06-19:
Well, that's weird. But easy fix then - I'll change from wget to curl.

ncosta24, 2024-06-19:
NICE!

Admonstrator, 2024-06-19:
Hopefully fixed in [0b78b6d](https://github.com/Admonstrator/glinet-tailscale-updater/commit/0b78b6dd974fdf6fca437d1dc8490eed86cd5210)

ncosta24, 2024-06-19:
[image]
[image]

ncosta24, 2024-06-19:
Thank you very much