admonstrator/glinet-tailscale-updater #6: Beryl GL-MT1300 Incorrect architecture

HarvsG, 2024-07-01:
```
root@GL-MT1300:~# sh update-tailscale.sh --ignore-free-space
[2024-07-01 13:47:39] [→] Checking for script updates
[2024-07-01 13:47:39] [✓] The script is up to date
┌────────────────────────────────────────────────────────────────────────┐
│ GL.iNet router script by Admon 🦭 for the GL.iNet community            │
| Version: 2024.06.20.01                                                 |
├────────────────────────────────────────────────────────────────────────┤
│ WARNING: THIS SCRIPT MIGHT POTENTIALLY HARM YOUR ROUTER!               │
│ It's only recommended to use this script if you know what you're doing.│
├────────────────────────────────────────────────────────────────────────┤
│ This script will update Tailscale on your router.                      │
│                                                                        │
│ Prerequisites:                                                         │
│ 1. At least 15 MB of free space.                                       │
│ 2. Firmware version 4 or higher.                                       │
│ 3. Architecture arm64, armv7 or mips.                                  │
└────────────────────────────────────────────────────────────────────────┘
[2024-07-01 13:47:40] [→] Checking if prerequisites are met
[2024-07-01 13:47:40] [✓] Firmware version: 4
[2024-07-01 13:47:40] [✓] Architecture: mips
[2024-07-01 13:47:40] [x] Not enough space available. Please free up some space and try again.
[2024-07-01 13:47:40] [x] The script needs at least 15 MB of free space. Available space: 9 MB
[2024-07-01 13:47:40] [x] If you want to continue, you can use --ignore-free-space to ignore this check.
[2024-07-01 13:47:40] [!] --ignore-free-space flag is used. Continuing without enough space
[2024-07-01 13:47:40] [!] Current available space: 9 MB
[2024-07-01 13:47:40] [!] xz is not installed. We can install it for you later.
[2024-07-01 13:47:40] [✓] Prerequisites are met.
┌──────────────────────────────────────────────────┐
| Are you sure you want to continue? (y/N)         |
└──────────────────────────────────────────────────┘
y
┌────────────────────────────────────────────────────────────────────────┐
│ WARNING: --ignore-free-space flag is used. This might potentially harm │
│ your router. Use it at your own risk.                                  │
│ You might need to reset your router to factory settings if something   │
│ goes wrong.                                                            │
└────────────────────────────────────────────────────────────────────────┘
┌──────────────────────────────────────────────────┐
| Are you sure you want to continue? (y/N)         |
└──────────────────────────────────────────────────┘
y
[2024-07-01 13:47:44] [→] Ok, continuing
[2024-07-01 13:47:44] [→] Detecting latest tiny tailscale version
/usr/sbin/tailscale: line 1: EL�4P4: not found
/usr/sbin/tailscale: line 2: syntax error: unexpected ")"
[2024-07-01 13:47:46] [→] The latest tailscale version is: 1.68.1
[2024-07-01 13:47:46] [→] Downloading latest tailscale version
[2024-07-01 13:47:47] [→] Creating backup of tailscale config
[2024-07-01 13:47:47] [✓] Backup created: /root/tailscale_config_backup/2024-07-01_13-47-47.tar.gz
[2024-07-01 13:47:47] [→] The binaries will not be backed up, you can restore them by using the --restore flag.
[2024-07-01 13:47:47] [→] Stopping tailscale
[2024-07-01 13:47:52] [→] Moving tailscale to /usr/sbin
[2024-07-01 13:48:31] [→] Removing temporary files
[2024-07-01 13:48:31] [→] Modifying gl_tailscale script to work with the new tailscale version
[2024-07-01 13:48:31] [✓] gl_tailscale script modified successfully
[2024-07-01 13:48:31] [→] Restarting tailscale
┌────────────────────────────────────────────────────────────────────────────────┐
| The update was successful. Do you want to make the installation permanent?     |
| This will make your tailscale installation persistent over firmware upgrades.  |
| Please note that this is not officially supported by GL.iNet.                  |
| It could lead to issues, even if not likely. Just keep that in mind.           |
| In worst case, you might need to remove the config from /etc/sysupgrade.conf   |
└────────────────────────────────────────────────────────────────────────────────┘
> Do you want to make the installation permanent? (y/N)
N
[2024-07-01 13:48:43] [✓] Script finished successfully. The new tailscale version (software, daemon) is:
/usr/sbin/tailscale: line 1: EL�4P4: not found
/usr/sbin/tailscale: line 2: syntax error: unexpected ")"
/usr/sbin/tailscaled: line 1: EL�4P4: not found
/usr/sbin/tailscaled: line 2: syntax error: unexpected ")"
```
I believe the issue is that it should use mipsel not mips

HarvsG, 2024-07-01:
NB I deleted the binaries before updating to make room

Admonstrator, 2024-07-01:
Could you please run `uname -a` on your device?

HarvsG, 2024-07-01:
Linux GL-MT1300 5.10.176 #0 SMP Sun Apr 9 12:27:46 2023 mips GNU/Linux

Admonstrator, 2024-07-01:
Unfortunately, the mipsel (mipsle) architecture isn't supported yet.
I am not sure how to get it working. The mipsle compiled version (see https://github.com/Admonstrator/glinet-tailscale-updater/releases/tag/v1.68.1 ) doesn't work.

Admonstrator, 2024-07-09:
The issue was fixed in 660fcc9f52adecb4542967c2a467ac30088541dd
Please re-run the script to update your Tailscale on Beryl.

closed 2024-07-31 by Admonstrator