admonstrator/glinet-tailscale-updater #7: Illegal Instruction

PaulW08, 2024-07-10:
BusyBox v1.35.0 (2023-04-09 12:27:46 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 22.03.4, r20123-38ccc47687
 -----------------------------------------------------
root@GL-E750:~# uname -a
Linux GL-E750 5.10.176 #0 Sun Apr 9 12:27:46 2023 mips GNU/Linux
root@GL-E750:~# wget -O update-tailscale.sh https://raw.githubuserconte
nt.com/Admonstrator/glinet-tailscale-updater/main/update-tailscale.sh &
& sh update-tailscale.sh
--2024-07-10 13:58:00--  https://raw.githubusercontent.com/Admonstrator/glinet-tailscale-updater/main/update-tailscale.sh
Resolving raw.githubusercontent.com... 185.199.111.133, 185.199.109.133, 185.199.110.133, ...
Connecting to raw.githubusercontent.com|185.199.111.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 26768 (26K) [text/plain]
Saving to: 'update-tailscale.sh'

update-tailscale. 100%[============>]  26.14K  --.-KB/s    in 0.01s   

2024-07-10 13:58:02 (2.45 MB/s) - 'update-tailscale.sh' saved [26768/26768]

[2024-07-10 13:58:02] [→] Checking for script updates
[2024-07-10 13:58:03] [✓] The script is up to date
┌────────────────────────────────────────────────────────────────────────┐
│ GL.iNet router script by Admon 🦭 for the GL.iNet community            │
| Version: 2024.07.09.03                                                 |
├────────────────────────────────────────────────────────────────────────┤
│ WARNING: THIS SCRIPT MIGHT POTENTIALLY HARM YOUR ROUTER!               │
│ It's only recommended to use this script if you know what you're doing.│
├────────────────────────────────────────────────────────────────────────┤
│ This script will update Tailscale on your router.                      │
│                                                                        │
│ Prerequisites:                                                         │
│ 1. At least 15 MB of free space.                                       │
│ 2. Firmware version 4 or higher.                                       │
│ 3. Architecture arm64, armv7 or mips.                                  │
└────────────────────────────────────────────────────────────────────────┘
[2024-07-10 13:58:03] [→] Checking if prerequisites are met
[2024-07-10 13:58:03] [✓] Firmware version: 4
[2024-07-10 13:58:03] [✓] Architecture: mips
[2024-07-10 13:58:03] [✓] Available space: 86 MB
[2024-07-10 13:58:03] [!] xz is not installed. We can install it for you later.
[2024-07-10 13:58:03] [✓] Prerequisites are met.
┌──────────────────────────────────────────────────┐
| Are you sure you want to continue? (y/N)         |
└──────────────────────────────────────────────────┘
y
[2024-07-10 13:58:06] [→] Detecting latest tiny tailscale version
Illegal instruction
[2024-07-10 13:58:13] [→] The latest tailscale version is: 1.68.2
[2024-07-10 13:58:13] [→] Downloading latest tailscale version
[2024-07-10 13:58:15] [→] Creating backup of tailscale config
tar: etc/config/tailscale: No such file or directory
tar: error exit delayed from previous errors
[2024-07-10 13:58:15] [✓] Backup created: /root/tailscale_config_backup/2024-07-10_13-58-15.tar.gz
[2024-07-10 13:58:15] [→] The binaries will not be backed up, you can restore them by using the --restore flag.
[2024-07-10 13:58:15] [→] Stopping tailscale
[2024-07-10 13:58:20] [→] Moving tailscale to /usr/sbin
[2024-07-10 13:58:20] [→] Removing temporary files
[2024-07-10 13:58:20] [→] Modifying gl_tailscale script to work with the new tailscale version
sed: /usr/bin/gl_tailscale: No such file or directory
[2024-07-10 13:58:20] [✓] gl_tailscale script modified successfully
[2024-07-10 13:58:20] [→] Restarting tailscale
┌────────────────────────────────────────────────────────────────────────────────┐
| The update was successful. Do you want to make the installation permanent?     |
| This will make your tailscale installation persistent over firmware upgrades.  |
| Please note that this is not officially supported by GL.iNet.                  |
| It could lead to issues, even if not likely. Just keep that in mind.           |
| In worst case, you might need to remove the config from /etc/sysupgrade.conf   |
└────────────────────────────────────────────────────────────────────────────────┘
> Do you want to make the installation permanent? (y/N)
y
[2024-07-10 13:58:34] [→] Making installation permanent
[2024-07-10 13:58:34] [→] Modifying /etc/sysupgrade.conf
[2024-07-10 13:58:34] [✓] Script finished successfully. The new tailscale version (software, daemon) is:
Illegal instruction
Illegal instruction
root@GL-E750:~# wget -O update-tailscale.sh https://raw.githubuserconte
nt.com/Admonstrator/glinet-tailscale-updater/main/update-tailscale.sh &
& sh update-tailscale.sh

PaulW08, 2024-07-10:
root@GL-E750:~# grep 'machine' /proc/cpuinfo | awk -F ': ' '{print $2}'
GL.iNet GL-E750
root@GL-E750:~# 
root@GL-E750:~# opkg update
Downloading https://fw.gl-inet.com/releases/v22.03.4/kmod-4.3.2/e750/Packages.gz
Updated list of available packages in /var/opkg-lists/glinet_core
Downloading https://fw.gl-inet.com/releases/v22.03.4/packages-4.2/ath79/glinet/Packages.gz
Updated list of available packages in /var/opkg-lists/glinet_gli_pub
Downloading https://fw.gl-inet.com/releases/v22.03.4/packages-4.2/ath79/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/glinet_gli_packages
Updating database.

Database update completed.
root@GL-E750:~# opkg install file
Installing file (5.41-1) to root...
Downloading https://fw.gl-inet.com/releases/v22.03.4/packages-4.2/ath79/packages/file_5.41-1_mips_24kc.ipk
Installing liblzma (5.4.1-1) to root...
Downloading https://fw.gl-inet.com/releases/v22.03.4/packages-4.2/ath79/packages/liblzma_5.4.1-1_mips_24kc.ipk
Installing libbz2-1.0 (1.0.8-1) to root...
Downloading https://fw.gl-inet.com/releases/v22.03.4/packages-4.2/ath79/packages/libbz2-1.0_1.0.8-1_mips_24kc.ipk
Installing libmagic (5.41-1) to root...
Downloading https://fw.gl-inet.com/releases/v22.03.4/packages-4.2/ath79/packages/libmagic_5.41-1_mips_24kc.ipk
Configuring libbz2-1.0.
Configuring liblzma.
Configuring libmagic.
Configuring file.
Updating database.
Database update completed.
root@GL-E750:~# file /bin/busybox
/bin/busybox: ELF 32-bit MSB executable, MIPS, MIPS32 rel2 version 1 (SYSV), dynamically linked, interpreter /lib/ld-musl-mips-sf.so.1, no section header

Admonstrator, 2024-07-10:
It seems that the architecture does not fit, even if it's MIPS.
Please try to run:

```
wget -O update-tailscale.sh https://raw.githubusercontent.com/Admonstrator/glinet-tailscale-updater/main/update-tailscale.sh && sh update-tailscale.sh --no-tiny
```

PaulW08, 2024-07-10:
I think that worked!On mobile now, so excuse any formatting issues.

root@GL-E750:~# wget -O update-tailscale.sh https://raw.githubusercontent.com/Admonstrator/glinet-tailscale-updater/main/update-tailscale.sh && sh u
pdate-tailscale.sh --no-tiny
--2024-07-10 16:28:42--  https://raw.githubusercontent.com/Admonstrator/glinet-tailscale-updater/main/update-tailscale.sh
Resolving raw.githubusercontent.com... 185.199.110.133, 185.199.111.133, 185.199.108.133, ...
Connecting to raw.githubusercontent.com|185.199.110.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 26768 (26K) [text/plain]
Saving to: 'update-tailscale.sh'

update-tailscale.sh   100%[========================>]  26.14K  --.-KB/s    in 0.007s  

2024-07-10 16:28:42 (3.70 MB/s) - 'update-tailscale.sh' saved [26768/26768]

[2024-07-10 16:28:42] [→] Checking for script updates
[2024-07-10 16:28:43] [✓] The script is up to date
┌────────────────────────────────────────────────────────────────────────┐
│ GL.iNet router script by Admon 🦭 for the GL.iNet community            │
| Version: 2024.07.09.03                                                 |
├────────────────────────────────────────────────────────────────────────┤
│ WARNING: THIS SCRIPT MIGHT POTENTIALLY HARM YOUR ROUTER!               │
│ It's only recommended to use this script if you know what you're doing.│
├────────────────────────────────────────────────────────────────────────┤
│ This script will update Tailscale on your router.                      │
│                                                                        │
│ Prerequisites:                                                         │
│ 1. At least 15 MB of free space.                                       │
│ 2. Firmware version 4 or higher.                                       │
│ 3. Architecture arm64, armv7 or mips.                                  │
└────────────────────────────────────────────────────────────────────────┘
[2024-07-10 16:28:43] [→] Checking if prerequisites are met
[2024-07-10 16:28:43] [✓] Firmware version: 4
[2024-07-10 16:28:43] [✓] Architecture: mips
[2024-07-10 16:28:43] [✓] Available space: 85 MB
[2024-07-10 16:28:43] [!] xz is not installed. We can install it for you later.
[2024-07-10 16:28:43] [✓] Prerequisites are met.
┌──────────────────────────────────────────────────┐
| Are you sure you want to continue? (y/N)         |
└──────────────────────────────────────────────────┘
y
[2024-07-10 16:28:45] [→] Detecting latest tailscale version
Illegal instruction
[2024-07-10 16:28:51] [→] The latest tailscale version is: tailscale_1.68.2_mips.tgz
[2024-07-10 16:28:51] [→] Downloading latest tailscale version
[2024-07-10 16:28:57] [→] Finding tailscale binaries in archive
[2024-07-10 16:29:14] [✓] Found tailscale binaries in: tailscale_1.68.2_mips
> Do you want to compress the binaries with UPX to save space? (y/N) y
[2024-07-10 16:29:16] [→] Ensuring xz is present and installing if necessary

[2024-07-10 16:31:34] [✓] xz is installed.
[2024-07-10 16:31:34] [→] Getting UPX
[2024-07-10 16:31:53] [→] Compressing tailscale with UPX
[2024-07-10 16:31:53] [→] This might take 2-3 minutes, depending on your router.
                       Ultimate Packer for eXecutables
                          Copyright (C) 1996 - 2024
UPX 4.2.4       Markus Oberhumer, Laszlo Molnar & John Reiser    May 9th 2024

        File size         Ratio      Format      Name
   --------------------   ------   -----------   -----------
tailscale  1/6  [................................................]  100.0%  | Command terminated by signal 9
7.96
tar: write error: No space left on device
[2024-07-10 16:32:14] [→] Compressing tailscaled with UPX
[2024-07-10 16:32:15] [→] This might take 2-3 minutes, depending on your router.
                       Ultimate Packer for eXecutables
                          Copyright (C) 1996 - 2024
UPX 4.2.4       Markus Oberhumer, Laszlo Molnar & John Reiser    May 9th 2024

        File size         Ratio      Format      Name
   --------------------   ------   -----------   -----------
Command terminated by signal 9
2.00
[2024-07-10 16:32:21] [→] Creating backup of tailscale config
tar: etc/config/tailscale: No such file or directory
tar: error exit delayed from previous errors
[2024-07-10 16:32:23] [✓] Backup created: /root/tailscale_config_backup/2024-07-10_16-32-21.tar.gz
[2024-07-10 16:32:23] [→] The binaries will not be backed up, you can restore them by using the --restore flag.
[2024-07-10 16:32:23] [→] Stopping tailscale
[2024-07-10 16:32:28] [→] Moving tailscale to /usr/sbin
[2024-07-10 16:32:54] [→] Removing temporary files
[2024-07-10 16:32:54] [→] Modifying gl_tailscale script to work with the new tailscale version
sed: /usr/bin/gl_tailscale: No such file or directory
[2024-07-10 16:32:54] [✓] gl_tailscale script modified successfully
[2024-07-10 16:32:54] [→] Restarting tailscale
┌────────────────────────────────────────────────────────────────────────────────┐
| The update was successful. Do you want to make the installation permanent?     |
| This will make your tailscale installation persistent over firmware upgrades.  |
| Please note that this is not officially supported by GL.iNet.                  |
| It could lead to issues, even if not likely. Just keep that in mind.           |
| In worst case, you might need to remove the config from /etc/sysupgrade.conf   |
└────────────────────────────────────────────────────────────────────────────────┘
> Do you want to make the installation permanent? (y/N)
[2024-07-10 16:32:54] [✓] Script finished successfully. The new tailscale version (software, daemon) is:
1.68.2
  tailscale commit: c3ed2ce8b57390b589914b9713cc64c2d3625412
  other commit: 789e6d2f8145ccfb04ca65c0de4d3d7c4b831ef8
  go version: go1.22.4
Segmentation fault

PaulW08, 2024-07-10:
Maybe not actually:

root@GL-E750:~# tailscale up
failed to connect to local tailscaled; it doesn't appear to be running

Admonstrator, 2024-07-12:
You might try to upgrade tailscale manually by downloading the correct version here:
https://pkgs.tailscale.com/stable/#static

closed 2024-07-31 by Admonstrator