admonstrator/glinet-tailscale-updater #8: issue with GL-E750 V2 / Mudi with this script

hetzbh, 2024-09-24:
Hi,
I tried your latest script (checked out and tested today) on my Mudi and I still get "Illegal instruction".

I then downloaded the mips version from the official tailscale package site ([here](https://pkgs.tailscale.com/stable/#static)). I downloaded and extracted the tgz file.

In this version, there are 2 separated files - tailscale and tailscaled, so all I had to do was to copy them to /usr/sbin directory (don't link them as this will not enable the user to run commands such "tailscale status").

Since then, everything works, specially since I ran your script and it updates the sysupgrade file to preserve the tailscale binaries when updating firmware.

So, my question: Could you update the script to use these packages please?

Admonstrator, 2024-09-24:
Thanks for the details - you can always run `--no-tiny` so the script will take the official binaries instead.
If you don't mind, give it a try.

hetzbh, 2024-09-26:
Well, I tried. 2 issues:

1. When it asks if I want to compress and use UPX and I specifically type "n" (instead of pressing Enter) - it tries to compress
2. After the install, the tailscale binary (not the tailscaled) gave "segmentation fault"
3. One small suggestion: when running this script, please stop tailscaled before doing anything. On Mudi, tailscaled + running the scripts kill the device in terms of usage (570% on tailscaled)

nolith, 2025-03-11:
Hello, I did receive my mudi v2 a few days ago and attempted a tailscale update today. 

Unfortunately it was not successful. 

The tiny version returns status code 127 with no output. 

The non tiny cannot be installed because of the lack of space. 

I did install manually the official binary by sending curl output into tar to expand the archive. 

Now I have some binaries but running tailscale status takes around 50seconds and overall it all seems not usable.

nolith, 2025-03-11:
I forgot to mention that the update scripts seems to do something with gl_tailscale that is not present on my mudi