admonstrator/glinet-tailscale-updater #9: Build latest tailscale version

reini-1, 2024-11-07:
It seems tailscale did not update releases to the latest tag:

```shell
curl -s "https://api.github.com/repos/tailscale/tailscale/releases" | jq -r 'sort_by(.published_at) | last | .tag_name'
v1.76.1
```

but the latest tag is newer and also the [tailscale changelog](https://tailscale.com/changelog) lists newer versions.

Maybe the built actions can use the latest tag:

```shell
curl -s "https://api.github.com/repos/tailscale/tailscale/git/refs/tags" | jq -r '[.[].ref | sub("^refs/tags/"; "") | select(test("^v*[0-9.]*$"))] | sort_by(sub("^v";"")|split(".")|map(tonumber)) | last'
v1.76.6
```

Admonstrator, 2024-11-07:
Seems that Tailscale did not create a release, they just created tags.
I don't know if it is wise to rely on tags instead of releases ... :-/

Admonstrator, 2024-11-07:
They updated the tailscale repo, so 1.76.6 is the newest release. My job already picked it up and created the correct tiny versions.
I would close this bug report because I think it's the best to use the official release from Tailscale instead of just the current tag.

https://github.com/Admonstrator/glinet-tailscale-updater/releases/tag/v1.76.6

closed 2024-11-07 by Admonstrator