admonstrator/glinet-tailscale-updater #13: Is 1.80.2 Available?

frenchpeacock, 2025-02-26:
First off, thank you so much for this code! You are a lifesaver!

My Tailscale admin portal shows me that my Flint 2 and Puli AX are both out of date, (they are both currently 1.80.0, and I see that the latest posted from Tailscale is 1.80.2). Am I posting something wrong with the code provided because it tells me I already have the latest version when I run it? Is this because 1.80.2 is not published in the official repository?

Thank you again, mad props!

Admonstrator, 2025-02-26:
Hey,

Unfortunately, Tailscale did not push the 1.80.2 to the repository yet - that's why the script isn't able to find it.

Admonstrator, 2025-02-26:
The script was changed to keep track of the tag releases in the tailscale repo.
1.80.2 is available now. 👍🏻

tswestendorp, 2025-02-26:
`versions.txt` is missing from the assets so (tiny) version detection fails

Admonstrator, 2025-02-27:
The issue was fixed, thanks everyone!

closed 2025-02-27 by Admonstrator